FarmBot/farmbot_os #669: Configurated advanced settings :: DNS nameserver bug

jsimmonds2, 2018-12-26:
Hi,
[ This forum thread ](https://forum.farmbot.org/t/ethernet-connection-issue/4286/14), `post #33`, describes the issue.
Thanks !

ConnorRigby, 2019-01-16:
@jsimmonds2 Can you give this a try on the latest beta when you get a chance. I _think_ i fixed it? not 100% sure though.

jsimmonds2, 2019-01-17:
Had already built `version 7.0.0` from commit `a7ea416b`  yesterday and that bug lives happily on.
The 5 files I had to change are the same ones as in `version 6.4.12`.   Here's the `git diff` from yesterday.
I would send in a PR except you seem to be refactoring the core structure of the project quite radically a.t.m.

[nameserver-naming-names-git-diff.txt](https://github.com/FarmBot/farmbot_os/files/2766803/z.txt)

B.t.w., where are the db migration sources ?   (I amended the generated one)

jsimmonds2, 2019-01-17:
Re: Try latest beta . . fresh clone ( up to commit 826d3fce ) :-1: 

`jhs@Comet:~/src/FarmBot/farmbot_os$ mix deps.get`
`* Getting nerves_wpa_supplicant (https://github.com/nerves-project/nerves_wpa_supplicant.git)`
`remote: Enumerating objects: 45, done.        `
`remote: Counting objects: 100% (45/45), done.        `
`remote: Compressing objects: 100% (35/35), done.        `
`remote: Total 464 (delta 15), reused 29 (delta 9), pack-reused 419        `
`Receiving objects: 100% (464/464), 129.61 KiB | 623.00 KiB/s, done.`
`Resolving deltas: 100% (234/234), done.`
**`fatal: reference is not a tree: 790e3ed33e65776834c6cdbe13174d02347657a1`**
`** (Mix) Command "git --git-dir=.git checkout --quiet 790e3ed33e65776834c6cdbe13174d02347657a1" failed`
`jhs@Comet:~/src/FarmBot/farmbot_os$ `

ConnorRigby, 2019-01-17:
Ah sorry about that. I assumed you used the images published to `releases`. I just fixed that in the last push.

jsimmonds2, 2019-01-17:
All good . . if you **really** think  I should still test this,  then I'll just use the latest 7.0.0 `IMG`  . . let me know  :electron:

jsimmonds2, 2019-01-19:
Hi @ConnorRigby,  I made enough time to test out that [v7.0.0-beta](https://github.com/FarmBot/farmbot_os/releases/download/v7.0.0-beta/farmbot-rpi3-7.0.0-beta.img) . . and same bug still exists.

( Then asked self  :  Why did I ever doubt that ❔ :-)

ConnorRigby, 2019-02-08:
https://github.com/FarmBot/farmbot_os/commit/70327210c4e57a67581dfb8877e0fcc2225d2228
I think this may fix the issue. There was a typo/key mismatch in the Farmbot code. There should be a new beta image coming up shortly.

jsimmonds2, 2019-02-08:
Thanks @ConnorRigby  . .  that minimal fix does the job !  Tested Ok.

ConnorRigby, 2019-02-09:
Thats great! Thanks for finding it.

closed 2019-03-05 by ConnorRigby