FarmBot/farmbot_os #674: NervesHub WebSocket client restarting due to 403, "Forbidden"

jsimmonds2, 2019-01-15:
I understand that NervesHub firmware distribution is very new . . so this issue may simply be a NervesHub Hosting issue.

My little RPi3B+ running FBOS 6.4.12 is getting very warm at 60.1 'C  (ambient is 34'C ! . . Australian Summer)
This seems to be due to too much CPU chewed by the BEAM restarting the WebSocket client every 6s.

`05:21:05.428 [error] Process #PID<0.27174.3> terminating`
`** (exit) {403, "Forbidden"}`
`    (stdlib) gen_fsm.erl:609: :gen_fsm.terminate/7`
`    (stdlib) proc_lib.erl:249: :proc_lib.init_p_do_apply/3`
`Initial Call: :websocket_client.init/1`
`Ancestors: [NervesHub.Socket, NervesHub.Supervisor, Farmbot.System.Supervisor, Farmbot,` `#PID<0.2930.0>]`
`Message Queue Length: 0`
`Messages: []`
`Links: [#PID<0.3169.0>]`
`Dictionary: []`
`Trapping Exits: false`
`Status: :running`
`Heap Size: 1598`
`Stack Size: 27`
`Reductions: 22662`
``
`05:21:11.728 [error] Process #PID<0.27185.3> terminating`

ConnorRigby, 2019-01-15:
60c isn't really too bad for rpi. It is considered "unsafe" at about 80-90. That said i think this is a good point. This should be on a back off timer probably as not to eat up cpu for no reason.

jsimmonds2, 2019-01-15:
Thanks . .  Intel XTU shows that my old Surface Pro 4 with Gen.7 i7 is capped to a package temp. of 90'C

ConnorRigby, 2019-01-15:
The path of least resistance here is to increase this value: https://github.com/FarmBot/farmbot_os/blob/staging/config/nerves_hub.exs#L8 to something a bit higher. Probably a minute or so.

The next option is to set a hook [here](https://github.com/FarmBot/farmbot_os/blob/staging/platform/target/nerves_hub_client.ex#L119-L126) and pattern match for the `forbidden` error, and after some amount of retries, just stop the `nerves_hub` application.

jsimmonds2, 2019-01-15:
Aaaaaahhhh . . I feel cooler now ! (TM)

ConnorRigby, 2019-01-22:
Should be fixed in v7.0.0!

closed 2019-01-22 by ConnorRigby