FarmBot/farmbot_os #685: Configurated Advanced settings :: some input boxes don't display on Windows 10 current Edge browser

jsimmonds2, 2019-02-12:
Had to use Chromium browser on Windows 10 to get my Farmbot 7.0.1 image configurated :)
Here's what I see using Microsoft Edge 
[image]

jsimmonds2, 2019-02-13:
Ooops, wrong repo !!    . . I'll send it to the Microsoft team  😉 🤣

ConnorRigby, 2019-03-05:
@jsimmonds2 we published a beta release today that should have a fix for this. I don't have easy access to a windows machine. Could you test it if you have a chance?

jsimmonds2, 2019-03-05:
@ConnorRigby 
Testing this will be my entertainment for the day . . and always happy to help out.

jsimmonds2, 2019-03-06:
Umm .. with 7.0.2-beta the problem still shows.

I'm not a competent browser-side developer but I'll have a crack at diagnosing the problem.
I s'pose it could be a bug in Edge, so I'll look around there first.

Here's Windows 10 Edge . . I noticed that the **IP METHOD** list element remains "highlighted"

[image]

and here's Chromium Version 72.0.3646 (64 bit)

[image]

jsimmonds2, 2019-03-06:
@ConnorRigby 
## Uh-oh, this may take me a while ™
I turned on Edge browser Developer Tools (F12) and then looked at the input box element for **IP ADDRESS**, it was suddenly rendered and displayed on the main window 😕   . . and then same happened for both the other missing inputs.

jsimmonds2, 2019-03-06:
@ConnorRigby,  I think my initial comment at the top was on the money.

Unless you have a paying Customer who **`Loves`** Windows I'm happy to close this issue.
Playing around (inspecting elements) with the Edge Developer Tools I can get all sorts of weirdness ->

[image]

ConnorRigby, 2019-03-06:
Ha! well leave it open, but we will make it a bit of a lower priority. We've been updating a few things in configurator lately anyway so taking a crack at Edge support wasn't really going well out of our way. We will investigate a little bit further before pushing another release

jsimmonds2, 2019-03-07:
@ConnorRigby ,  at the same time, I'll try for a little reproducer that I can hit the Microsoft Edge Support team with.
MSFT _seem_ to be genuine about wanting Edge to be compliant against current standards and expected behaviours !   Amazes me how many browsers we still have that Devs may need to cope with !
Thanks.

jsimmonds2, 2019-04-12:
**_Maybe_**, just maybe, MSFT's next Edge fixes this one 🤞 (Microsoft Edge based on Chromium !!)

[image]

ConnorRigby, 2019-07-09:
@jsimmonds2 have you tried this on V8? i've fixed some html syntax errors, and added some metadata in the `head` field that i think should fix this.

jsimmonds2, 2019-08-22:
Latest Beta from Windows Insider program of Edge browser (based on Chromium) fixes everything !

closed 2019-08-22 by jsimmonds2