FarmBot/farmbot_os #686: Latest 7.0.2-beta crashes early in boot :: esqlite3_nif.so

jsimmonds2, 2019-03-17:
Hoping to try out Elixir 1.8.1 and latest FarmBot OS bits so I grabbed the 3 day-old beta release from [https://github.com/FarmBot/farmbot_os/releases/download/v7.0.2-beta/farmbot-rpi3-7.0.2-beta.img](https://github.com/FarmBot/farmbot_os/releases/download/v7.0.2-beta/farmbot-rpi3-7.0.2-beta.img)

B.t.w. , where is the SHA256sum for released IMG files ?

This is as far as boot gets :

[image]

ConnorRigby, 2019-03-25:
Yeah this is a bug while upgrading to Elixir 1.8. It has to do with some incompatibilities with the `pc` rebar3 plugin and `elixir_make`

ConnorRigby, 2019-03-25:
> B.t.w. , where is the SHA256sum for released IMG files

This is a good idea. I'll add it to our CI deployment

ConnorRigby, 2019-03-25:
Should be fixed with https://github.com/FarmBot/farmbot_os/commit/c4407486a628b4aa768281a8a9324881a5bcfb15

closed 2019-03-25 by ConnorRigby

jsimmonds2, 2019-03-26:
Err, . .  nup.   Either that commit is not in the published `7.0.2-beta` IMG or it's no good.

Thanks for the SHA256sum !

I can wait 'til Eight  (v8.0.0) 👍

ConnorRigby, 2019-03-26:
> I can wait 'til Eight (v8.0.0)

You and me both

Anyway the beta that came up had cached assets in it from CI. I busted the cache and it's rebuilding right now.

jsimmonds2, 2019-03-26:
> Anyway the beta that came up had cached assets in it from CI. I busted the cache and it's rebuilding right now.

That's fixed it !   Thanks.