FarmBot/farmbot_os #697: The network configuration occurs n times, depending in the waiting for the interface loop.

alde103, 2019-04-04:
I believe that there is a bug in the init function in [manager](https://github.com/FarmBot/farmbot_os/blob/staging/platform/target/network/manager.ex#L36).

The configuration of the network is performed n times, depending on how many loops are made during the wait for an up interface loop.

ConnorRigby, 2019-04-05:
Ah so this doesn't configure network `n` times. It's used as a blocking method to wait for the iface to come up. Basically if the iface doesn't exist yet (in the case of Linux hasn't loaded the driver yet, it just recurses after waiting a second. This is essentially a remnant of before SystemRegistry existed.

alde103, 2019-04-05:
A quick fix maybe:

def wait_until_up(interface) do
      unless interface in Nerves.NetworkInterface.interfaces() do
         Process.sleep(1000)
         wait_until_up(interface)
      end
end

and just call it instead of the [unless](https://github.com/FarmBot/farmbot_os/blob/staging/platform/target/network/manager.ex#L39) block.

alde103, 2019-04-05:
A quick fix maybe a `wait_until_up` funtion and just call it instead of the [unless](https://github.com/FarmBot/farmbot_os/blob/staging/platform/target/network/manager.ex#L39) block.

 ```
defp wait_until_up(interface) do
    unless interface in Nerves.NetworkInterface.interfaces() do
      Process.sleep(1000)
      wait_until_up(interface)
    end
end
```

ConnorRigby, 2019-07-03:
Fixed in #760

closed 2019-07-03 by ConnorRigby