FarmBot/farmbot_os #710: Provide Farmware name to Farmware executable

avonwyss, 2019-05-17:
The Farmware needs to know the name under which the configs can be retrieved (e.g. the prefix used in the environment values for config entries). This prefix (farmware name) could therefore also be made available in the environment, for instance as `FARMWARE_NAME`, in [lib/farmbot/farmware/runtime.ex](https://github.com/FarmBot/farmbot_os/blob/7ca4e5d936ff7ebbbbf2daa4d5755925e6394fd7/lib/farmbot/farmware/runtime.ex#L118)

ConnorRigby, 2019-05-17:
The farmware name is hardcoded it the manifest.json file. If you want to know the name it dynamicly you could read this file and pull it from there

closed 2019-05-17 by ConnorRigby

avonwyss, 2019-05-17:
@ConnorRigby How can I read the manifest file used from within the Farmware?