FarmBot/farmbot_os #722: v8.0.0 Configurator :: static IPv4 address bug

jsimmonds2, 2019-05-29:
As mentioned to @ConnorRigby in a FarmBot Forum PM, there's a simple bug which prevents legitimate `static` IPv4 address network configurations from being created.

Here's my `git diff` :

[image]

ConnorRigby, 2019-05-29:
Thanks for pointing this out. I'm actually in the process of helping get the top secret new nerves networking library https://github.com/nerves-networking/vintage_net ready for use. Which is why I haven't spent much time working on the network stack for v8 as it is rather temporary

jsimmonds2, 2019-05-29:
Ok, thanks.  Last question : will the Configurator AP handling also change ?  (therefore, should I still hunt out the bug(s) in relevant current v8.0.0 code in `next` branch ?)

ConnorRigby, 2019-05-30:
Yep AP handling will be moved into the same system.

jsimmonds2, 2019-06-01:
Cool !   (and extra cool if **`static`** IPv4 adresses are supported in the official release too :)
(You and Frank are having way too much fun :):)

closed 2019-06-01 by jsimmonds2

ConnorRigby, 2019-07-09:
Should have been fixed in #760