FarmBot/farmbot_os #762: Pre-release :: busybox dep. issue in v8.0.0-rc0 / staging

jsimmonds2, 2019-07-02:
Sorry for this bit of silly noise, but just wanted to make sure I can build this version before grabbing the pre-built artefact IMG file.

Building inside a fresh clone.

[image]

ConnorRigby, 2019-07-02:
Did you try cleaning and rebuilding? It looks like the download was corrupted

jsimmonds2, 2019-07-02:
Well, I haven't yet because I saw that the hash compare failed on **`busybox-1.30.1.tar.bz2`** and not the version that Frank has kindly provided :)
P.s. surely a corrupt download would be detected way earlier ?

jsimmonds2, 2019-07-03:
Did another clean clone (of v8.0.0-rc1 this time) and get same failure in `mix firmware` stage.
Can't see how my (very simple) environment could be a cause . . I've tried both `MIX_ENV=dev` and `MIX_ENV=prod`

[image]

ConnorRigby, 2019-07-03:
Can you open an issue on nerves-networking/elixir_busybox

jsimmonds2, 2019-07-03:
Of course . . thanks :)

fhunleth, 2019-07-08:
@jsimmonds2 Thanks for the report and for filing a bug over at the `elixir_busybox` repo.

@ConnorRigby `busybox` is an optional dependency. I was thinking that you wouldn't need it for farmbot_os since you'd be enabling the required busybox packages in your custom systems. (Not that this fixes any issue with `elixir_busybox`, that is...)

ConnorRigby, 2019-07-08:
Yeah I'm only using the Elixir BusyBox dep for now to save me time building systems as I have 3 custom ones that take quite a while to build/release. I will switch back at some point in the near future.

Also figured I'd stress test the new BusyBox package :troll:

jsimmonds2, 2019-07-09:
@fhunleth  thanks so much for your ever-so-patient encouragement !

Removing that `:busybox` dep from `farmbot_os` `v8.0.0-rc18` gives me a clean build :+1:  Yay.
(B.t.w. that dep. isn't flagged `optional` in the `farmbot_os` config _ :puzzled: )

Now I can go sleuthing the issue that's causing `:nerves_runtime :stopped` in this pre-release version !

jsimmonds2, 2019-07-09:
For the record, here's a snip of the RingLogger log showing `:nerves_runtime` stopping.

[image]

closed 2019-07-09 by jsimmonds2