FarmBot/farmbot_os #792: [Sequence] Execute a sequence on boot

ConnorRigby, 2019-07-09:
This would be a nice feature, and may be useful for `home on boot` type situations

RickCarlino, 2019-07-09:
IDEA BIN: We could add this to the device table to ensure referential integrity, prevent user from accidentally deleting their boot sequence, etc...

closed 2019-11-07 by ConnorRigby

jsimmonds2, 2022-04-08:
There is a "User Disappointer" bug in the solution submitted in [PR#1487](https://github.com/FarmBot/farmbot_os/pull/1487)

Any User's valid and pre-existing `Boot Sequence` will fail to run ( once ), immediately after an **`FBOS Soft Reset`** ( a.k.a. Factory Reset ) with a failure like this ( ignore the Erlang OOM . . seems something to do with the `log_attach` done earlier )

[image]

As is evident, **`Auto Sync`** takes longer on **_first boot_** after a `Soft Reset` than it normally would.

I will have a code fix PR ready in next few days.

PR #1052 (2019-11-07): Sequence on boot
Fixes #792