FarmBot/farmbot_os #799: [Sync] [Init] Flag to force a factory reset

ConnorRigby, 2019-07-09:
While helping debug [this](https://forum.farmbot.org/t/farmbot-os-disconnects-and-reconnects/3910/4) issue, it came to my attention that Farmbot can get into a situation where it cannot connect to AMQP but has a valid token and such. If a user would like to reset their bot, they can't since AMQP is not connected. 

It would be a good feature to have that a user can `POST` to some end point that would force the next token fetch to return a 4xx code to FarmbotOS user-agents.

ConnorRigby, 2019-12-19:
Fixed in #1090

closed 2019-12-19 by ConnorRigby

roryaronson, 2020-01-15:
From Rick: "Ah, I think this one was QA'ed only from an FBOS perspective- there is currently no UI element to toggle the flag, so I will need add a UI for it first."