FarmBot/farmbot_os #885: [Farmware] Installation errors for manifest with config

gabrielburnworth, 2019-07-24:
# FarmBot OS version
`8.0.0-rc50`

# Errors encountered
```
Farmware failed to download Farmware manifest
errors: [
  config: {"invalid config item", []},
  config: {"invalid config item", []}
]
```
```
protocol Jason.Encoder not implemented for %FarmbotCore.Asset.FarmwareInstallation.Manifest.Config
```

# To reproduce
Install via the following manifest:
```
{
  "package": "config test",
  "language": "sh",
  "author": "FarmBot, Inc.",
  "description": "test",
  "farmware_manifest_version": "2.0.0",
  "package_version": "0.0.0",
  "farmbot_os_version_requirement": ">= 8.0.0-any",
  "farmware_tools_version_requirement": ">= 0.0.0",
  "url": "",
  "zip": "https://github.com/FarmBot-Labs/farmware_manifests/archive/master.zip",
  "executable": "sh",
  "args": "-c 'echo config test'",
  "config": {
    "1": {"name": "config_1", "label": "Config 1", "value": 1},
    "2": {"name": "config_2", "label": "Config 2", "value": "two"}
  }
}
```

# Relevant code
https://github.com/FarmBot/farmbot_os/blob/staging/farmbot_core/lib/farmbot_core/asset/farmware_installation/manifest.ex

ConnorRigby, 2019-07-25:
@gabrielburnworth the error message wasn't very clear here. the config fields are missing a key called `order`, tho i don't remember why that was added. Do you remember if it is necessary?

closed 2019-07-25 by ConnorRigby

gabrielburnworth, 2019-07-25:
Farmware are installing and configs appear to be working properly after this change.

The number key identifying each config in the `config` object replaces the function of the `order` field.

PR #888 (2019-07-25): Bug/farmware config render
Fixes #885