FarmBot/farmbot_os #897: [CeleryScript] Implement `set_pin_io_mode`

gabrielburnworth, 2019-07-29:
# FarmBot OS Version
`8.0.0-rc53`

# Previous implementation
https://github.com/FarmBot/farmbot_os/blob/1fe52021c71bfe068c2be09bd8ded99c211804f6/lib/farmbot/firmware/uart_handler/uart_handler.ex#L384-L387

https://github.com/FarmBot/farmbot_os/blob/1fe52021c71bfe068c2be09bd8ded99c211804f6/lib/farmbot/firmware/utils.ex#L27-L29

ConnorRigby, 2019-08-08:
Is `set_pin_mode` ever used? is it an RPC? I didn't add it to V8 because i couldn't find any usage of it.

ConnorRigby, 2019-08-08:
Just kidding i found the old RPC. Reimplementing

gabrielburnworth, 2019-08-20:
errors:
```
function FarmbotCeleryScript.SysCalls.set_pin_io_mode/2 is undefined or private
```
```
no function clause matching in FarmbotOS.SysCalls.SetPinIOMode.extract_set_pin_mode/1
```
```
STUB HANDLER: unknown code: {"32", {:pin_mode_write, [p: 13, m: 0]}}
```
Here is the implementation used in production:
```JSON
{
    "kind": "set_pin_io_mode",
    "args": {
        "pin_number": 13,
        "pin_io_mode": 0
    }
}
```

https://github.com/FarmBot/farmbot_os/blob/1fe52021c71bfe068c2be09bd8ded99c211804f6/lib/farmbot/celery_script/ast/arg/pin_io_mode.ex#L1-L12

PR #901 (2019-08-08): Enhancement/set pin io mode
Fixes  #897
PR #922 (2019-08-21): Fix set_pin_io mode
Fixes #897
closed 2019-08-21 by ConnorRigby