FarmBot/farmbot_os #899: upload fails for images larger than ~3MB

gabrielburnworth, 2019-07-29:
# Error
v7:
```
Failed to upload /tmp/images/1539365725.jpg %Farmbot.HTTP.Response{body: "Metadata part is too large.", headers: [{"X-GUploader-UploadID", "<REDACTED>"}, {"Content-Length", "27"}, {"Date", "Fri, 12 Oct 2018 17:35:39 GMT"}, {"Server", "UploadServer"}, {"Content-Type", "text/html; charset=UTF-8"}, {"Alt-Svc", "quic=\":443\"; ma=2592000; v=\"44,43,39,35\""}], status_code: 400}. Going to retry.
```
v8:
```
Failed to upload image: {:ok, %Tesla.Env{__client__: %Tesla.Client{adapter: nil, fun: nil, post: [], pre: [{Tesla.Middleware.BaseUrl, :call, ["https://storage.googleapis.com/<REDACTED>/"]}, {Tesla.Middleware.Headers, :call, [[{"user-agent", "FarmbotOS/8.0.0-rc53 (rpi3) rpi3 ()"}]]}, {Tesla.Middleware.FormUrlencoded, :call, [[]]}]}, __module__: FarmbotExt.API, body: "Metadata part is too large.", headers: [{"date", "Fri, 12 Oct 2018 17:35:39 GMT"}, {"server", "UploadServer"}, {"content-length", "27"}, {"content-type", "text/html; charset=UTF-8"}, {"x-guploader-uploadid", "<REDACTED>"}, {"alt-svc", "quic=\":443\"; ma=2592000; v=\"44,43,39,35\""}], method: :post, opts: [], query: [], status: 400, url: "https://storage.googleapis.com/<REDACTED>/"}}
```

ConnorRigby, 2019-08-08:
I believe this to be a backend error. 4XX http error means it was rejected by the receiving end. The only thing Farmbot can do to help this is clean up the error message

closed 2019-08-08 by ConnorRigby

gabrielburnworth, 2019-08-08:
There is a possibility that something gets malformed in large image upload requests (which may have a solution in FBOS), but we would need more debug information to verify. In any case, larger than 3MB images still fail to upload, so where is the appropriate place to keep this issue open?

ConnorRigby, 2019-08-08:
I think it would need to be transfered to the API as it's not a farmbot specific error. Anything trying to use this endpoint would have this issue

PR #900 (2019-08-08): Clean up error message for uploading images
Fixes #899