FarmBot/farmbot_os #909: Feature  - Proxy Awareness

txmit1, 2019-08-10:
Would be great for corporate  and educational installations to see some ability of proxy  awareness ability either PAC address  configuration and username and password or just http proxy in the farmbot setup for those behind corperate firewalls. Our biggest let down  currently for getting farmbot fully functional. 

I know WpaEpa just was released in v7 that has helped initially  but those  who use EPA typically  have  proxies to deal with too.

ConnorRigby, 2019-08-12:
This has been on my short list of features to add. Will bump priority on it and hopefully get something working soon.

txmit1, 2019-08-12:
Amazing, love the work. We're looking forward to it will save us many many
work around

Much Appreciated
Tim Mitchell
B.Science (Chemistry/Applied Mathematics)
G. Dip Education (Secondary)

adunwell, 2019-11-18:
Very Interested in this feature addition, in an EDU space behind a proxy also. Following for Update

ConnorRigby, 2019-11-19:
I've looked into this a small amount, but haven't come to any full conclusions yet. The problem i see up front is that "proxy" as a term is kind of ambiguous. Farmbot OS uses many protocols including but not limited to (off the top of my head)

* NTP
* HTTP (S)
* AMQP (over tls)

so the issue with having a "proxy" is it would need to handle all of those protocols, not just HTTP to work correctly. On top of that, Farmbot does not control the client implementation of any of those protocols, it uses libraries to speak those protocols so to even have a remote chance of supporting proxying, the farmbot team would need to fork each library and add proxy support to it. 

the other solution might be to just proxy the pure TCP and UDP connections at the lowest level. I don't know of any tooling that does this personally, but it might be out there. I have to research more. This seems like the most reasonable way to solve the problem, but i'm not 100% sure it's a real thing or not.