FarmBot/farmbot_os #1009: `move_abs` block no longer honors `speed` settings

RickCarlino, 2019-09-28:
We've received reports from multiple customers that the speed setting on the `move_abs` block is being ignored. I was able to reproduce this morning on a device running `8.1.0-rc3`.

gabrielburnworth, 2019-09-28:
# Relevant code

v7: `G00 X0.0 Y0.0 Z0.0 A500.0 B500.0 C500.0` https://github.com/FarmBot/farmbot_os/blob/679a2a187f2d2e47d394e8575b7e1db418fba7ce/lib/farmbot/firmware/uart_handler/uart_handler.ex#L311-L316

v8: `G00 X0.00 Y0.00 Z0.00 S100.00` https://github.com/FarmBot/farmbot_os/blob/e727d1b533cc54818f1f708078481331973514f1/farmbot_os/lib/farmbot_os/sys_calls.ex#L233

firmware: https://github.com/FarmBot/farmbot-arduino-firmware/blob/293440baa4ffbf18e5772a36e06927ac0ff6b908/src/G00Handler.cpp#L48

ConnorRigby, 2019-10-10:
Fixed in #1010

closed 2019-10-10 by ConnorRigby