FarmBot/farmbot_os #1022: [GPIO] Should Pi GPIO 4 be available for PinBinding ?

jsimmonds2, 2019-10-07:
As mentioned over in [this Farmbot Forum post](https://forum.farmbot.org/t/using-farmbotos-beta-updates/3951/65), I'm seeing issues using GPIO 4  ( on RPi3B+ ).

Here's what I'm seeing on current FarmotOS Release **`v8.0.4`** probing with latest **`raspi-gpio`** utility . .

[image]

ConnorRigby, 2019-10-07:
Not exactly sure what would cause gpio 4 not to work correctly. I don't think it has any special meaning on any pi model. 

Are you using the Farmbot gpio hat? It shouldn't be required anymore but ive not actually verified that.

jsimmonds2, 2019-10-07:
That J8 40-pin header on my **`RPi3B+`** is connected to nothing but an FTDI cable on pins 6,8,10.

The [reference RPi3B+ reduced schematic](https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_3bplus_1p0_reduced.pdf) ****does**** tag pin 7 (GPIO4) as a **`General Purpose Clock`** (??)

ConnorRigby, 2019-10-10:
@jsimmonds2 could you try this on 8f4f7d2ea190bd7067b2b74656e827084489b778 or later please? I think i may have accidentally fixed it by enabling pull downs in software. If that doesn't work, we may just need to remove this pin from the available pins list.

jsimmonds2, 2019-10-11:
@ConnorRigby , that's made a change, for sure, but . . on Very First Sync

* Pi GPIO 4 ( **`unconnected physical header pin 7`** ) **`triggers`** (??)
* Pi GPIO 18 ( **`unconnected physical header pin 12`** ) **`is not triggered`**

I'll keep investigating, etc

[image]

ConnorRigby, 2019-10-11:
I think gpio-4 might just not work with rpi3 b+ unfortunately. I don't have one to test with i don't think but it looks like we should remove it from the list of allowed pins.

jsimmonds2, 2019-10-12:
> not work with rpi3 b+ unfortunately. I don't have one to test with

That's sad.  Isn't Rpi3B+ the platform in the newest 'bots ?

jsimmonds2, 2019-10-16:
Before any removing happens, I need to actually **`test`** that the pin "won't work" :)
I was only observing the console logs evident after 1st Sync.

FBOS initializes all pins available for binding to **`input`** with **`pulldown`** ,  is that correct ?

ConnorRigby, 2019-10-16:
> That's sad. Isn't Rpi3B+ the platform in the newest 'bots ?

We still ship with the standard Rpi3 currently.

> FBOS initializes all pins available for binding to **`input`** with **`pulldown`** , is that correct ?

only pins that have an associated binding are initialized, but yes

ConnorRigby, 2019-10-31:
I'm going to close this for now since it doesn't effect official kits. If we ship the next version with the rpi3b+, i will reopen

closed 2019-10-31 by ConnorRigby