FarmBot/farmbot_os #1034: [Bug] no compiler for home

jsimmonds2, 2019-10-24:
# Expected Behavior
On Controls page, press Home button (house icon)
No error occurs and bot homes correctly

# Actual Behavior
On FarmbotOS Console, a Runtime error is logged.  I don't have a real bot but I imagine nothing moved.

# Steps to Reproduce
Install latest FarmbotOS Release from **`.IMG`** downloaded from GH ( **`v8.1.0`** at **`4b12dfaf`** ) and use **`my.farm.bot`**  WebApp Release ( says **`VERSION: FCE9ACD7`** )

# Results of `git log -1`

Run `git log -1` and paste the results here.
**`n/a`**

Picture this :)

[image]

ConnorRigby, 2019-10-24:
Thanks @jsimmonds2. Pushing a fix right now

ConnorRigby, 2019-10-31:
Fixed with 29bff5750110cab60701d22c84d32a5785f5da3f

closed 2019-10-31 by ConnorRigby