FarmBot/farmbot_os #1040: [Bug] SchedulerUsageReporter not reporting actual CPU usage

jsimmonds2, 2019-10-31:
Testing on **`v8.2.0-rc3`**

WebApp // Device CPU Usage (now MicroStateAccounting view of Scheduler's efforts) is stuck at
### 0%

Found two bugs which I'll send in as a PR soon.

ConnorRigby, 2019-10-31:
We knew about this one but kept it at CPU usage for legacy reasons. Most folks aren't concerned with what erlangs schedulers are, and measuring actual CPU usage is pretty pointless because almost all of the CPU is dedicated to Erlang. Similarly with memory usage. We are measuring stats from beam because it basically is our OS but the names of the values were kept normalized

jsimmonds2, 2019-10-31:
I do understand what you've just said about the names and how the new summary percentage is exactly the **work** that two classes of scheduler are doing ( versus **sleeping** :)

There are, however, bugs in that new code.  If you examine the raw values coming from **`:msacc`** you might agree.   I'll send in a PR to show you.

closed 2019-11-04 by ConnorRigby