FarmBot/farmbot_os #1056: [Bug] Can't enable Firmware Debug log

jsimmonds2, 2019-11-08:
Testing on Pre-Release **`v8.3.0-rc3`**

Expected `fbos_config.firmware_debug_log` to change as the FE Log Setting item was toggled.

Picture this :)

[image]

gabrielburnworth, 2019-11-08:
Try `fbos_config.arduino_debug_messages`.

jsimmonds2, 2019-11-08:
Oh, Ok and **thanks**. 
Sorry for the noise.

Some refactoring going on that I hadn't bothered to follow through :(

jsimmonds2, 2019-11-08:
A deeper look tells me that bug(s) lurk.

E.g [https://github.com/FarmBot/farmbot_os/blob/staging/farmbot_core/lib/farmbot_core/firmware_side_effects.ex](https://github.com/FarmBot/farmbot_os/blob/staging/farmbot_core/lib/farmbot_core/firmware_side_effects.ex#L155)
```
  should_log? = Asset.fbos_config().firmware_debug_log
```
:-)

ConnorRigby, 2019-11-08:
I've noticed a bug lately where I have to refresh the page  too make that panel work

jsimmonds2, 2019-11-08:
Yes, same. That toggle slider seems to have some busy code underneath it !

jsimmonds2, 2019-11-21:
Finally **got** my Firmware Debug logs :)

* Built a **`farmduino_k14.hex`** with debug logging enabled.
* Forced **`firmware_debug_log`** to true in **`FbosConfig`** using
```
FarmbotCore.Asset.Private.mark_dirty! FarmbotCore.Asset.fbos_config(), %{firmware_debug_log: true}

```
Surely **`arduino_debug_messages`** is now/nearly obsolete ?

ConnorRigby, 2019-11-22:
i believe `arduino_debug_messages` and `firmware_debug_log` are the same thing. There may be a bug where `botstate.arduino_debug_messages` is not updated when `firmware_debug_log` is updated and vice versa.

ConnorRigby, 2019-11-22:
oh actually, i see that both of those values are real values, not one as a mirror of the other. i'll have to cross reference which one is **supposed** to be used for this feature.

PR #1070 (2019-11-26): Fix firmware_debug_logs and arduino_debug_messages being ignored
Don't know which of these params is supposed to be used for this
feature, so added both of them.

Fixes #1056
closed 2019-11-26 by ConnorRigby