FarmBot/farmbot_os #1066: Farmbot Connection Issue

derletztename, 2019-11-16:
I tried to connect the farmbot (OS 8.2.2) with the newest version of the WebApp and failed. WebApp seems to run just fine. But for reasons I don't understand the Farmbot won't connect to the WebApp. Maybe someone could help me figuring out what's the issue. As both systems are just freshly installed, it might be a bug in the current farmbotOS code?

WebApp sais:

> web_1          | Started GET "/api/public_key" for 192.168.8.239 at 2019-11-16 15:05:46 +0000
> web_1          | Processing by Api::PublicKeysController#show as JSON
> web_1          |   Parameters: {"public_key"=>{}}
> web_1          |   Rendering text template
> web_1          |   Rendered text template (Duration: 0.0ms | Allocations: 1)
> web_1          | Completed 200 OK in 1ms (Views: 0.5ms | ActiveRecord: 0.0ms | Allocations: 267)
> web_1          | 192.168.8.239 - - [16/Nov/2019:15:05:46 UTC] "GET /api/public_key HTTP/1.1" 200 451
> web_1          | - -> /api/public_key
> web_1          | Started POST "/api/tokens" for 192.168.8.239 at 2019-11-16 15:05:46 +0000
> web_1          | Processing by Api::TokensController#create as JSON
> web_1          |   Parameters: {"user"=>{"credentials"=>"[FILTERED]"}, "token"=>{"user"=>{"credentials"=>"[FILTERED]"}}}
> web_1          | [active_model_serializers] Rendered ActiveModel::Serializer::Null with ActiveSupport::HashWithIndifferentAccess (0.11ms)
> web_1          | Completed 422 Unprocessable Entity in 73ms (Views: 1.1ms | ActiveRecord: 1.0ms | Allocations: 940)
> web_1          | 192.168.8.239 - - [16/Nov/2019:15:05:46 UTC] "POST /api/tokens HTTP/1.1" 422 33
> web_1          | - -> /api/tokens

And on farmbot side, when I connect directly via browser:

> MatchError at GET /network
> no match of right hand side value: []
> platform/target/configurator/vintage_network_layer.ex
> No code available.
> anonymous fn/1 in FarmbotOS.Platform.Target.Configurator.VintageNetworkLayer.list_interfaces/0farmbot
>  Show only app frames
> •	 platform/target/configurator/vintage_network_layer.ex:10anonymous fn/1 in FarmbotOS.Platform.Target.Configurator.VintageNetworkLayer.list_interfaces/0
> •	 elixir lib/enum.ex:1336Enum."-map/2-lists^map/1-0-"/2
> •	 elixir lib/enum.ex:1336Enum."-map/2-lists^map/1-0-"/2
> •	 lib/farmbot_os/configurator/router.ex:63anonymous fn/2 in FarmbotOS.Configurator.Router.do_match/4
> •	 lib/plug/router.ex:259FarmbotOS.Configurator.Router.dispatch/2
> •	 lib/farmbot_os/configurator/router.ex:1FarmbotOS.Configurator.Router.plug_builder_call/2
> •	 lib/plug/debugger.ex:122FarmbotOS.Configurator.Router.call/2
> •	 plug_cowboy lib/plug/cowboy/handler.ex:12Plug.Cowboy.Handler.init/2
> Request info
> URI:
> http://192.168.8.239:80/network
> Query string:
> Headers
> accept
> text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3
> accept-encoding
> gzip, deflate
> accept-language
> de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
> connection
> keep-alive
> cookie
> _farmbot_session=wqUaSzvvw3CIrh+Q1wJ0FVsykWGt987bSHjKwvzSnee4F7wURlmtvLPr7vPZhhkHPidoc65ynZKAdRli7tW98n7TvXDbjEFgpgeKqXThr1ksPjEa+OGMA22FP7Fsatmm
> host
> 192.168.8.239
> upgrade-insecure-requests
> 1
> user-agent
> Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/78.0.3904.97 Safari/537.36
> Session
> auth_config_email
> "barmsen@hotmail.de"
> auth_config_password
> "permabot01"
> auth_config_server
> "http://192.168.8.152:3000"
> ifname
> "wlan0"
> iftype
> "wireless"
> net_config_dns_name
> nil
> net_config_domain
> nil
> net_config_identity
> nil
> net_config_ipv4_address
> "0.0.0.0"
> net_config_ipv4_gateway
> "0.0.0.0"
> net_config_ipv4_method
> "dhcp"
> net_config_ipv4_subnet_mask
> "255.255.0.0"
> net_config_name_servers
> nil
> net_config_ntp1
> nil
> net_config_ntp2
> nil
> net_config_password
> nil
> net_config_psk
> "permabot01"
> net_config_reg_domain
> "US"
> net_config_security
> "WPA-PSK"
> net_config_ssh_key
> nil
> net_config_ssid
> "GliNet"

RickCarlino, 2019-11-16:
@derletztename What is the output of `git log -1` from within the Web App directory?

derletztename, 2019-11-17:
paul@permabotserver:~/Farmbot-Web-App$ git log -1
commit b8867cc16571e653427d3f578389ead00dc7a5ff (HEAD -> master)
Merge: 8eaf643 0e2c016
Author: Rick Carlino <RickCarlino@users.noreply.github.com>
Date:   Thu Nov 14 08:06:21 2019 -0600

    Merge pull request #1573 from FarmBot/thursday_deploy
    
    v8.2.0 - Iridescent Iris
paul@permabotserver:~/Farmbot-Web-App$

ConnorRigby, 2019-11-18:
@derletztename what version if the raspberry pi are you using? Farmbot doesn't seem to detect any network devices which could mean a power issue or possibly a broken raspberry pi.

derletztename, 2019-11-19:
I am using a Raspberry Pi 3 B+.
I tested another one at home, just the Raspi (no Farmduino) with a freshly installed version of farmbot os 8.2.2. It's the same issue, but with another IP adress. So the webapp recognizes the farmbot/ Rasps somehow, but only receives "Unprocessable Entitys" ?

> web_1          | Started POST "/api/tokens" for 192.168.178.65 at 2019-11-19 18:42:41 +0000
> web_1          | Processing by Api::TokensController#create as JSON
> web_1          |   Parameters: {"user"=>{"credentials"=>"[FILTERED]"}, "token"=>{"user"=>{"credentials"=>"[FILTERED]"}}}
> web_1          | [active_model_serializers] Rendered ActiveModel::Serializer::Null with ActiveSupport::HashWithIndifferentAccess (0.12ms)
> web_1          | Completed 422 Unprocessable Entity in 81ms (Views: 1.3ms | ActiveRecord: 1.1ms | Allocations: 940)
> web_1          | 192.168.178.65 - - [19/Nov/2019:18:42:41 UTC] "POST /api/tokens HTTP/1.1" 422 33
> web_1          | - -> /api/tokens

derletztename, 2019-11-19:
Maybe somethings off with my .env? But most things are turned off there anyways. I am very confused as I couldn't have done so much wrong at this point of setting up the farmbot. Most of it is fresh out of the box downloaded from github

ConnorRigby, 2019-11-22:
I'm sort of confused about your environment. Why are you accessing configurator at `192.168.8.239` instead of it's captive portal access point?

derletztename, 2019-11-22:
I am very sorry, for not even understanding your question properly. I do not access configurator to this address, it gets it from the Web App. I just installed everything fresh out of the box and tried to get it working.
But what did the trick was going into logger and seeing the actual error. For reasons I can't understand, I mistyped the password over and over again. Or maybe before there were other reasons, I also installed 8.2.3 this time and now there was the password error.
I am just very happy that I made progress. Thx for pointing me in the right direction :)

ConnorRigby, 2019-12-02:
Glad to see you got it sorted out!

closed 2019-12-02 by ConnorRigby