FarmBot/farmbot_os #1074: Incorrect units in Firmware Parameter log message

jsimmonds2, 2019-11-28:
Changing MAX SPEED (MM/S) for X, Y or Z axis results in a Log Message which append units as **`(steps/mm)`**.  Surely that's not a unit of speed ?
 
Picture this :)

[image]

ConnorRigby, 2019-12-02:
This log actually is correct. The app does calculations based on the value stored in `steps per mm` to convert the data to `mm per second`. So the user inputs `100 mm / s` and farmbot receives `55 steps per second`

closed 2019-12-02 by ConnorRigby

jsimmonds2, 2019-12-02:
Understand about that calculation, just that there is a **different** control for STEPS PER MM.

> and farmbot receives `55 steps per second`

That would be 550 and as you said, the unit really is **steps per second**  :)