FarmBot/farmbot_os #1105: [bug] {X,Y,Z} Axis movement timeout reports from firmware cause FBOS Factory Reset

jsimmonds2, 2020-01-08:
[This Forum topic](https://forum.farmbot.org/t/farmbot-factory-reset-after-update-to-9-0-0/5758/22) describes the symptoms.

The root problem seems to be in **FirmwareReset** somewhere,  trying to reset the serial device.
Here's a RingLogger snippet showing **:farmbot_core** app exiting.

[image]

RickCarlino, 2020-01-08:
Thanks for the bug report @jsimmonds2 ! You are two steps ahead of us, it seems. Rory and I noticed this one yesterday- I will be investigating today.

jsimmonds2, 2020-01-09:
Here's more detail from the serial console.
The firmware ( my case **farmduino_k14** `6.4.2.G` commit `1711db1d` )

Looks like the firmware sent up a **79** which is not yet handled.

[image]

RickCarlino, 2020-01-09:
Nice find @jsimmonds2. Fairly certain the first issue is patched up. Will take a look into this one next.

RickCarlino, 2020-01-09:
@jsimmonds2 Can you try out `v9.0.1-rc3` and see if the issue persists?

jsimmonds2, 2020-01-09:
@RickCarlino , as soon as AmazonAWS lets me have it, I'll fire up on **v9.0.2** Release.  Thanks !

RickCarlino, 2020-01-10:
How's it looking, @jsimmonds2 ? My device seems to be chugging along after an OTA from 9.0.0 => 9.0.2. Rory noted a "hiccup" of some sort that he could not isolate and which went away after a reboot. Not ideal, but I think that is better than the factory reset issues reported in 9.0.0.

jsimmonds2, 2020-01-10:
As luck would have it, Amazon AWS currently has some network path issues delivering the v9.0.2 .IMG file to little old me down in AU !

I'll keep on retrying . . also, my attempt to update OTA also failed to run ( quietly ) . . maybe related ?

Thanks for your **all-nighter** to get a fix out !

jsimmonds2, 2020-01-10:
@RickCarlino , up now on **v9.0.2** . . no more crashes if **FIND HOME {X,Y,Z}** time out,
but **CALIBRATE {X,Y,Z}** seem to hang forever.

In my case ( just a Raspberry Pi 3 and a Farmduino V1.4 ; no other hardware ) I have **TIMEOUT AFTER (SECONDS)** set to 10, so it looks like the Axis movement timeout is getting swallowed and not reported.

Can you test CALIBRATION on a real bot with a shortened movement timeout value ?

jsimmonds2, 2020-01-10:
> Can you test CALIBRATION on a real bot with a shortened movement timeout value ?

@RickCarlino , maybe that's a "never mind" . . I'll go read the firmware code.
Looks like the **Farmduino** firmware handles **CALIBRATE** autonomously . . . ( I was **sure** that command timed out before your v9.0.2 fixes :-| )

RickCarlino, 2020-01-10:
Thanks for the updates @jsimmonds2. It seems that you are not the only person experiencing this issue. I did most of my testing on v1.2 (RAMPS) hardware, so it's possible this problem is exclusive to the Farmduino. Details to follow.