FarmBot/farmbot_os #1110: [bug, minor] CI commit hash not found on GH ?

jsimmonds2, 2020-01-10:
I love traceability in software changes.

Probably my myopia, but for FBOS **v9.0.2** from the CI artefacts, the commit hash can't be found in the GH **farmbot_os** repo. (?)

Here is the CI **.IMG** running right now ( 5a95d6c3 ).
I can't see that commit on GitHub.  ( help me :)

[image]

RickCarlino, 2020-01-10:
@jsimmonds2 Can you try again- I am seeing it on Github fine right now. Maybe it was related to those CDN hiccups you mentioned last night?

jsimmonds2, 2020-01-10:
Thanks @RickCarlino  !  User-error ( mine ). Seems I need to check the **master** branch for CI builds :)

closed 2020-01-10 by jsimmonds2