FarmBot/farmbot_os #1111: Firmware never recovers from reboot

RickCarlino, 2020-01-10:
# Steps to Reproduce
 * Create a sequence that uses the firmware (`move_abs`, `calibrate`, etc...)
 * Run the sequence to ensure that your firmware is running OK
 * Reboot the MCU from the Web App (not FBOS, just the MCU) and give it some time to reboot itself.
 * Run the sequence again.

# Expected Behavior

Firmware should be available after a reboot.

# Actual Behavior

Firmware is not available and requires a power cycle to fix. The following logs are observed after multiple attempts to run the firmware:

```

[info]  Calibrating all axes
[error] CeleryScript error [#Reference<0.30652697.825753601.95251>]:
        "Firmware error: :transport_boot"
[info]  Failed to execute command: Firmware error: :transport_boot
[info]  Moving to (0, 0, 0)
[info]  Movement failed. Retrying up to 3 more time(s)
[info]  Movement failed. Retrying up to 2 more time(s)
[info]  Movement failed. Retrying up to 1 more time(s)
[error] CeleryScript error [#Reference<0.30652697.825753601.150841>]:
        "movement error(s): :transport_boot, :transport_boot,
        :transport_boot, :transport_boot"
[info]  Failed to execute command: movement error(s):
        :transport_boot, :transport_boot, :transport_boot,
        :transport_boot
[info]  Moving to (0, 0, 0)
[info]  Movement failed. Retrying up to 3 more time(s)
[info]  Movement failed. Retrying up to 2 more time(s)
[info]  Movement failed. Retrying up to 1 more time(s)
[error] CeleryScript error [#Reference<0.30652697.825753601.154476>]:
        "movement error(s): :transport_boot, :transport_boot,
        :transport_boot, :transport_boot"
[info]  Failed to execute command: movement error(s):
        :transport_boot, :transport_boot, :transport_boot,
        :transport_boot
[info]  Moving to (0, 0, 0)
[info]  Movement failed. Retrying up to 3 more time(s)
[info]  Movement failed. Retrying up to 2 more time(s)
[info]  Movement failed. Retrying up to 1 more time(s)
[error] CeleryScript error [#Reference<0.30652697.825753601.158343>]:
        "movement error(s): :transport_boot, :transport_boot,
        :transport_boot, :transport_boot"
[info]  Failed to execute command: movement error(s):
        :transport_boot, :transport_boot, :transport_boot,
        :transport_boot
```

RickCarlino, 2020-01-10:
# Workaround

The issue is resolvable by emergency locking / unlocking the device.

jsimmonds2, 2020-01-10:
Confirm same issue observed yesterday on mine.  Used same ESTOP, UNLOCK workaround :)

RickCarlino, 2020-01-14:
I am fairly certain [this line](https://github.com/FarmBot/farmbot_os/blob/staging/farmbot_firmware/lib/farmbot_firmware.ex#L244) is setting the wrong default. Will investigate tomorrow.

RickCarlino, 2020-05-01:
Has anyone seen this recently? I believe this was fixed in 9.2.2...

jsimmonds2, 2020-05-04:
Not seen . . with no attempt to reproduce on **v9.2.2** 😝

closed 2020-05-04 by RickCarlino