FarmBot/farmbot_os #1112: Sequence Queueing Issues

RickCarlino, 2020-01-13:
Forum user `dmbgo` states that executing a sequence while one is already running causes unexpected behavior. [Link to original forum post](https://forum.farmbot.org/t/farmbot-factory-reset-after-update-to-9-0-0/5758/41?u=rickcarlino).

jsimmonds2, 2020-01-15:
@RickCarlino I had a play with this.
The only way that I found to make it happen is by using the `TEST` button on the **Sequences** page, e.g.
* Select first Sequence, press `TEST`
* **quickly** Select a different Sequence, press `TEST`

RickCarlino, 2020-01-15:
Thanks so much for doing the research, @jsimmonds2 ! This is 100% still a bug, but it is good to know that it is confined to edge cases.

 > Do the Sequences Use-Cases ( Stories :) allow for Canceling a running Sequence ?

This is possible via `e-stop`. Did you have a  different use case in mind?

 > Do we need to "serialize" Sequence requests ? . . or "cancel" all running ones if a new one arrives ?

It's been a while since I've looked into this part of the app, and its not how I typically use my device so I will need to give it a look. I have a hunch that we designed queueing as the intended default behavior, if memory serves me correct. Will need to investigate further. Details to follow

mdingena, 2021-03-09:
I can confirm this happens when running two sequences from the `Run` button in the Sequence designer.

I was running (via `Run` button) a sequence that takes photos of the entire garden bed. While it was doing that for a bit, I tapped `Run` on another sequence that scans the bed for weeds.

The expected behaviour was that it would either perform that sequence after finishing the current running one, OR it would abort the current running one immediately to start the triggered one.

The observed behaviour was that FarmBot now started ping-ponging between coordinates and taking normal photos and plant detection photos (and getting them mixed up in the process).