FarmBot/farmbot_os #1114: [bug, minor] Recurrent FarmEvent causes too many Logger records

jsimmonds2, 2020-01-15:
I set up 1 Event to run 1 Sequence per Day until :infinity :).
This is what appears in the RingLogger output . . can these 15 second "Developer" logs be shut off ?
### **:-)**

[image]

RickCarlino, 2020-01-15:
@jsimmonds2 That's an "Everything is OK" alarm from the FarmEvent runner :stuck_out_tongue_winking_eye:. I can see how the log is useful when debugging the FarmEvent runner (especially in situations where the runner silently dies), but do agree it is not very useful when the FarmEvent runner is working as intended. I might just comment this out, since we have not needed to debug this part of FBOS frequently. I can un-comment the line in dev later if needed.

RickCarlino, 2020-01-15:
I've commented it out in 03b7b4d

commit 03b7b4d4 (2020-01-15): Reduce verbosity of scheduler logs. Fixes #1114