FarmBot/farmbot_os #1117: [defects] Farmware page for 3rd-party Farmwares

jsimmonds2, 2020-01-20:
This may not be the correct repository for this report; I don't know WebApp code base well enough to decide.

As mentioned in [this FarmBot Forum post](https://forum.farmbot.org/t/farmware-moveabsolute-and-executesequence-not-working/5784/11) and following 2 replies, there are problems evident after **INSTALL**-ing and then **REMOVE**-ing Customer-written Farmware packages.

* After `INSTALL`-ing ( successfully ) a user-written Farmware package, the temporary Farmware name used during the installation process remains in the Farmware list.
* Removing that temporary Farmware name also ( silently ) deletes the actual user-written package also, leaving a Farmware name in the list which cannot be removed ( error Farmware not found ) !

Pic below, show the situation after a good install of **wc-farmware** package

[image]

RickCarlino, 2020-01-20:
Thanks for the detailed report, @jsimmonds2 . I am getting settled in to the office and will reply to your forum posts momentarily.

jsimmonds2, 2020-01-27:
** **UPDATE** **

That `Unknown Farmware 502 (pending install)` situation was my User Error ;; the URL inside the Manifest file did not match the URL I used for the ( successful ) installation !

There's still a couple of issues though.
* A little while ( 30s ? ) after a fresh installation, the `Information` panel shows a red `COULD NOT FETCH PACKAGE NAME` sub-panel which shows a `RETRY` button.  See image below.
* After `REMOVE`-ing my Farmware, nothing changes on the Farmware page !  ( my Farmware still seems to exist ! )

[image]

closed 2021-05-14 by RickCarlino