FarmBot/farmbot_os #1128: 30 second timeout on Farmware

whitecapsO, 2020-01-27:
(see https://forum.farmbot.org/t/farmware-moveabsolute-and-executesequence-not-working/5784/28)

From JSSimmonds: "this new Farmware behavior was introduced by @connor on Dec-14 2019 . . allows any Farmware to run for only 30s max. The code needs a re-work ( and a unit test )."

To recreate the behaviour:
Install this Farmware: https://raw.githubusercontent.com/whitecapsO/hello-farmware-test/master/manifest.json

It runs this code: https://github.com/whitecapsO/hello-farmware-test/blob/master/hello.py

The code tries to log 5 messages after the Hello Farmware message. It then tries to make 5 simple x & y movements.

On my Farmbot I only get 3 messages logged (big pauses between each message) before the Farmware stops i.e.
Hello Farmware
Message 1 Test
Message 2 Test

I also removed all the log writes, to test if just the 5 move_absolutes work. Only 3 moves work.

RickCarlino, 2020-01-27:
Thanks for the detailed report @whitecapsO . I'm still working on some fixes for the Express bots but have this on my radar as my next action item thereafter.

jsimmonds2, 2020-01-27:
@whitecapsO , you got the "wrong" Connor in your post. He's **ConnorRigby** on GitHub :)

RickCarlino, 2020-01-27:
...and it's perhaps best for us to not "@" him unnecessarily now that he is no longer at Farmbot. Although we still do talk regularly, I'd prefer not to fill his inbox with FarmBot-related emails.

RickCarlino, 2020-01-28:
**UPDATE:** I have added a warning message for other developers who author long running farmwares. It will be released with FBOS v9.1.0, which is expected to be out in less than a week.

https://github.com/FarmBot/farmbot_os/compare/qa/arduino-firmware?expand=1#diff-11b756f456bbcc43f104d46ce8bcc718R58-R66

whitecapsO, 2020-01-28:
Sure thing re Connor. Thanks for adding the message :).

jsimmonds2, 2020-01-28:
> I have added a warning message for other developers who author long running farmwares

Hoping that's just a temporary `BAND-AID®` ?
30s ain't very long running . . just enough for 2 or 3 simple statements !

There's a 10s delay **between** statements which is also worrying . . that's a huge amount of wasted time ( the running Farmware has the entire platform for its 'exclusive' use ).  I'll have a look at this.

jsimmonds2, 2020-01-28:
@whitecapsO , can you edit your OP to remove that (at)connor ?  It's referring to some other GitHub user who presumably is puzzled by the "mention"  email coming from GitHub (!)

whitecapsO, 2020-01-28:
Done.

jsimmonds2, 2020-01-28:
> Done.

Nope.  I meant your initial post for **this Issue** :)

RickCarlino, 2020-01-28:
> Hoping that's just a temporary BAND-AID® ?
 > Connor was having problems with Farmware running amok

Hopefully it is just temporary. As I remember, there were some very difficult
tradeoffs that happened with the creation of the v8 CeleryScript runtime.

 > the running Farmware has the entire platform for its 'exclusive' use

I believe this was the main reason for adding the timeout- there were bug
 reports of farmwares causing the system to hang if they were not cleanly exited
 (eg: runaway loops, infinite `sleep()` calls, etc..)

closed 2021-05-14 by RickCarlino