FarmBot/farmbot_os #1225: Unused variable warning is polluting the Serial Console :)  10.1.3

jsimmonds2, 2020-07-26:
[image]

RickCarlino, 2020-07-26:
I take the blame for this one @jsimmonds2 . While implementing the filter groups / group iteration feature, I was having issues with a part of the CSVM and worked around it temporarily. Definitely a bug, but the warning does not point to a significant problem, rather it points to some corner cutting on my part. I hope I can fix this in the upcoming v3 MOVE TO block.

jsimmonds2, 2020-07-27:
> the blame
"blame" . . What's _that_ ? :-)

Can CircleCI catch the real **stderr** stream ? Or **compile time** warnings like this ?

RickCarlino, 2020-07-27:
:stuck_out_tongue: I need to do a patch release today anyway to un-bork an unrelated issue. I will include a simple fix for this in 10.1.4.

jsimmonds2, 2020-07-28:
Ooops . . got a New one now ( only on Serial Console **stderr** , not SSH ) 10.1.4 commit **1c5ef14b**

[image]

jsimmonds2, 2020-08-04:
Hello @RickCarlino ! Has this one ( above ) been cleaned up yet ? :)  Thanks !

RickCarlino, 2020-08-05:
@jsimmonds2 It's pretty low priority given the other things that are going on this week, but I will need to go in there next week when writing the V3 MOVE block that's coming up.

Did the fix make it worse? I can remove it if that's the case.

jsimmonds2, 2020-08-05:
Thanks @RickCarlino .  No problems 😅  It's **not** worse after the recent fix, just a **different** unused variable.

Rarely use the Console device, but it's the only way I know of to see the **stdio** and **stderr** streams of the BEAM.

RickCarlino, 2020-08-10:
@jsimmonds2 Found a fix while working on unrelated features. The fix is in bb077b13 and will be deployed as part of v11.0.0

closed 2020-08-10 by RickCarlino

jsimmonds2, 2020-08-11:
### 👍

commit bb077b13 (2020-08-10): [UNSTABLE] Remove unused variable warning. Fixes #1225