FarmBot/farmbot_os #1357: Can't flash Arduino with firmware

gueux, 2021-06-18:
Hi, i have some problems with installing firmware to Arduino Mega through FarmBotOS.

Arduino connected to RaspberryPi3 cable and green LED onto RAMP board started to flash.
Also, as i see Raspberry succesfully connected to WebApp and MQTT:

[image]

but when I try to flash firmware, the error "Flash firmware failed" appears:

[image]

I've connected to Raspberry3 via Serial Console and got next error in terminal:

```
iex(farmbot@farmbot-000000008aa873b0.local)3> warning: CeleryScript Error: {:noproc, {GenServer, :call, [FarmbotCore.Firmware.UARTCore, {:flash_firmware, "arduino"}, 600000]}}
  (elixir 1.10.4) lib/gen_server.ex:1013: GenServer.call/3
  (farmbot_celery_script 14.1.2) lib/farmbot_celery_script/sys_calls.ex:372: FarmbotCeleryScript.SysCalls.ok_or_error/3
  (farmbot_celery_script 14.1.2) lib/farmbot_celery_script/step_runner.ex:40: FarmbotCeleryScript.StepRunner.execute/3
  (farmbot_celery_script 14.1.2) lib/farmbot_celery_script/step_runner.ex:15: FarmbotCeleryScript.StepRunner.step/3
```

Could you suggest whats wrong and how i can flash the proper firmware to Arduino?

RickCarlino, 2021-06-18:
@gueux You are running an outdated version of FBOS. Older versions of FBOS 14.x had some firmware issues that were recently fixed. Please upgrade and let me know if that fixes the problem.

You can upgrade by reflashing FBOS, or by running `sudo docker-compose run rake releases:publish` from your server's shell and then upgrading from inside of the web app.

gueux, 2021-06-19:
@RickCarlino i've updated farmbot os to the latest 14.1.3-rc2 version, but unfortunately issue is the same one. 

Thats what i get when click onto FlashFirmware: 
```
iex(farmbot@farmbot-000000008aa873b0.local)3> RingLogger.attach
:ok

13:52:17.976 [info]  CeleryScript OK
iex(farmbot@farmbot-000000008aa873b0.local)4> warning: CeleryScript Error: {:noproc, {GenServer, :call, [FarmbotCore.Firmware.UARTCore, {:flash_firmware, "arduino"}, 600000]}}
  (elixir 1.10.4) lib/gen_server.ex:1013: GenServer.call/3
  (farmbot_celery_script 14.1.3-rc2) lib/farmbot_celery_script/sys_calls.ex:372: FarmbotCeleryScript.SysCalls.ok_or_error/3
  (farmbot_celery_script 14.1.3-rc2) lib/farmbot_celery_script/step_runner.ex:40: FarmbotCeleryScript.StepRunner.execute/3
  (farmbot_celery_script 14.1.3-rc2) lib/farmbot_celery_script/step_runner.ex:15: FarmbotCeleryScript.StepRunner.step/3

13:52:25.145 [error] CeleryScript error [#Reference<0.2907558172.537657345.28763>]: "{:noproc, {GenServer, :call, [FarmbotCore.Firmware.UARTCore, {:flash_firmware, \"arduino\"}, 600000]}}"

13:52:25.246 [info]  Failed to execute command: {:noproc, {GenServer, :call, [FarmbotCore.Firmware.UARTCore, {:flash_firmware, "arduino"}, 600000]}}

13:52:25.247 [error] CeleryScript error [#Reference<0.2907558172.537657345.28763>]: "{:noproc, {GenServer, :call, [FarmbotCore.Firmware.UARTCore, {:flash_firmware, \"arduino\"}, 600000]}}"
```

And thats what happened when click onto RestartFirmware:
```
13:53:32.754 [info]  CeleryScript OK
iex(farmbot@farmbot-000000008aa873b0.local)4> warning: CeleryScript Exception:
  :erlang.send(FarmbotCore.Firmware.UARTCore, :reset_state)
  (farmbot_core 14.1.3-rc2) lib/firmware/uart_core.ex:80: FarmbotCore.Firmware.UARTCore.restart_firmware/1
  (farmbot 14.1.3-rc2) lib/farmbot_os/sys_calls.ex:195: FarmbotOS.SysCalls.firmware_reboot/0
  (farmbot_celery_script 14.1.3-rc2) lib/farmbot_celery_script/sys_calls.ex:372: FarmbotCeleryScript.SysCalls.ok_or_error/3
  (farmbot_celery_script 14.1.3-rc2) lib/farmbot_celery_script/step_runner.ex:40: FarmbotCeleryScript.StepRunner.execute/3
  (farmbot_celery_script 14.1.3-rc2) lib/farmbot_celery_script/step_runner.ex:15: FarmbotCeleryScript.StepRunner.step/3

13:53:36.955 [error] CeleryScript error [#Reference<0.2907558172.537657345.62551>]: "argument error"

13:53:37.042 [info]  Failed to execute command: argument error

13:53:37.042 [error] CeleryScript error [#Reference<0.2907558172.537657345.62551>]: "argument error"
```

BTW, i can manually compile and upload latest firmware through Arduino IDE, so seems board and usb cable is suitable.

gueux, 2021-06-19:
After some of arduino restart this error appears:

```
14:35:26.806 [info]  Rebooting inactive Farmduino. 2859697

14:35:26.931 [info]  Starting firmware flash.

14:35:27.081 [info]  Using default reset function

14:35:27.097 [debug] Writing firmware to MCU... ["-patmega2560", "-cwiring", "-P/dev/ttyAMA0", "-b115200", "-D", "-V", "-v", "-Uflash:w:/srv/erlang/lib/farmbot_core-14.1.3-rc2/priv/arduino_firmware.hex:i"]

14:35:27.200 [info]  Begin flash attempt 1
iex(farmbot@farmbot-000000008aa873b0.local)17>
nil

14:35:51.626 [info]  Attempt 1 failed.

14:35:51.748 [info]  "\navrdude: Version 6.1-svn-20130917, compiled on May 28 2021 at 14:31:52\n         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/\n         Copyright (c) 2007-2009 Joerg Wunsch\n\n         System wide configuration file is \"/etc/avrdude.conf\"\n         User configuration file is \"/root/.avrduderc\"\n         User configuration file does not exist or is not a regular file, skipping\n\n         Using Port                    : /dev/ttyAMA0\n         Using Programmer              : wiring\n         Overriding Baud Rate          : 115200\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_getsync(): timeout communicating with programmer\n\navrdude done.  Thank you.\n\n"

14:35:51.870 [info]  Begin flash attempt 2
```

gueux, 2021-06-19:
@RickCarlino  Seems i've found the reason of this problem. FarmbotOS use the wrong port definition for avrdude programmator tool.

```
iex(farmbot@farmbot-000000008aa873b0.local)5> {:ok, fun} = FarmbotCore.Firmware.Resetter.find_reset_fun("")
{:ok, #Function<0.91348444/0 in FarmbotCore.Firmware.Resetter.use_default/0>}
iex(farmbot@farmbot-000000008aa873b0.local)6> Avrdude.flash("/srv/erlang/lib/farmbot_core-14.1.3-rc2/priv/arduino_firmware.hex", "/dev/ttyAMA0", fun)
{"\navrdude: Version 6.1-svn-20130917, compiled on May 28 2021 at 14:31:52\n         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/\n         Copyright (c) 2007-2009 Joerg Wunsch\n\n         System wide configuration file is \"/etc/avrdude.conf\"\n         User configuration file is \"/root/.avrduderc\"\n         User configuration file does not exist or is not a regular file, skipping\n\n         Using Port                    : /dev/ttyAMA0\n         Using Programmer              : wiring\n         Overriding Baud Rate          : 115200\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_ReceiveMessage(): timeout\navrdude: stk500v2_getsync(): timeout communicating with programmer\n\navrdude done.  Thank you.\n\n",
 1}
```

```
iex(farmbot@farmbot-000000008aa873b0.local)9> {:ok, fun} = FarmbotCore.Firmware.Resetter.find_reset_fun("")
{:ok, #Function<0.91348444/0 in FarmbotCore.Firmware.Resetter.use_default/0>}
iex(farmbot@farmbot-000000008aa873b0.local)10> Avrdude.flash("/srv/erlang/lib/farmbot_core-14.1.3-rc2/priv/arduino_firmware.hex", "/dev/ttyUSB0", fun)
{"\navrdude: Version 6.1-svn-20130917, compiled on May 28 2021 at 14:31:52\n         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/\n         Copyright (c) 2007-2009 Joerg Wunsch\n\n         System wide configuration file is \"/etc/avrdude.conf\"\n         User configuration file is \"/root/.avrduderc\"\n         User configuration file does not exist or is not a regular file, skipping\n\n         Using Port                    : /dev/ttyUSB0\n         Using Programmer              : wiring\n         Overriding Baud Rate          : 115200\n         AVR Part                      : ATmega2560\n         Chip Erase delay              : 9000 us\n         PAGEL                         : PD7\n         BS2                           : PA0\n         RESET disposition             : dedicated\n         RETRY pulse                   : SCK\n         serial program mode           : yes\n         parallel program mode         : yes\n         Timeout                       : 200\n         StabDelay                     : 100\n         CmdexeDelay                   : 25\n         SyncLoops                     : 32\n         ByteDelay                     : 0\n         PollIndex                     : 3\n         PollValue                     : 0x53\n         Memory Detail                 :\n\n                                  Block Poll               Page                       Polled\n           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack\n           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------\n           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00\n           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00\n           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00\n           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00\n           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00\n           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00\n           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00\n           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00\n\n         Programmer Type : Wiring\n         Description     : Wiring\n         Programmer Model: AVRISP\n         Hardware Version: 15\n         Firmware Version Master : 2.10\n         Vtarget         : 0.0 V\n         SCK period      : 260.5 us\n\navrdude: AVR device initialized and ready to accept instructions\n\nReading | ################################################## | 100% 0.02s\n\navrdude: Device signature = 0x1e9801\navrdude: reading input file \"/srv/erlang/lib/farmbot_core-14.1.3-rc2/priv/arduino_firmware.hex\"\navrdude: writing flash (48968 bytes):\n\nWriting | ################################################## | 100% 7.96s\n\navrdude: 48968 bytes of flash written\n\navrdude done.  Thank you.\n\n",
 0}
```

Seems there is no fallback port option for trying both ports  `/dev/ttyAMA0` and `/dev/ttyUSB0`

RickCarlino, 2021-06-19:
@gueux I will assume you are using an Arduino that you did not purchase from FarmBot, Inc. If you are using non-official hardware, you need to manually set the firmware path.

Here is how you can manually set your UART device path (`ttyUSB0`):

0. If you do not know your Arduino's device path, reboot the device. It will provide a log that tells you all available UART device paths (usually ttyAMA0 or ttyUSB0).
1. Enable ADVANCED SETTINGS.
2. Change your Arduino path to `ttyUSB0` or whatever the correct path is.

---
[image]
---
[image]

closed 2021-06-19 by RickCarlino