FarmBot/farmbot_os #1454: fail to falsh fambot _os

alkhlama, 2021-12-07:
[image]
[image]
[image]

RickCarlino, 2021-12-07:
[image]
@alkhlama You need to select a device model.
https://software.farm.bot/v13/FarmBot-Software/getting-started.html#farmbot-model

closed 2021-12-07 by RickCarlino

alkhlama, 2021-12-07:
It is a custom device

[image]<https://user-images.githubusercontent.com/1388608/145083916-3463e1f5-9030-4051-8a1d-2d9b621b9c56.png>

https://software.farm.bot/v13/FarmBot-Software/getting-started.html#farmbot-model

RickCarlino, 2021-12-07:
@alkhlama You still need to select a device type. `None` is not a valid choice. If it is an Arduino Mega, then you can select Genesis v1.2.

alkhlama, 2021-12-08:
Greetings
I use mks gen 1.4 board instead of arduino and ramps 1.4  Combination how can I Configure it in the code

RickCarlino, 2021-12-08:
We do not support that board. For anyone who finds this issue via search, there is an additional discussion [here](https://github.com/FarmBot/farmbot-arduino-firmware/issues/163).