FarmBot/farmbot_os #1474: WiFi Adapters

sabas1080, 2022-01-26:
Hi

Can I install other Wi-Fi network adapters?

Thanks

RickCarlino, 2022-01-26:
@sabas1080 The answer to that question is usually "yes". When we build the Linux kernel for FarmBot OS, we try to add as many USB network peripherals as possible. With that being said, we can't guarantee that every USB adapter will work. Many Express customers have successfully purchased off-the-shelf USB Ethernet adapters for this purpose.

If you have not purchased a FarmBot yet and have a Raspberry Pi available, you can verify compatibility by flashing FBOS to the Pi (no need to attach peripherals) and see if configurator succeeds with your chosen adapter.

sabas1080, 2022-01-26:
@RickCarlino do you have a list of compatible adapters or chips?

RickCarlino, 2022-01-26:
@sabas1080 I do not, but if there's an adapter you had in mind (specifically, the chipset inside) I can look it up. We use buildroot and our buildroot config contains a list of all the drivers available: https://github.com/FarmBot/farmbot_system_rpi3/blob/main/linux-5.10.defconfig

Also, if you are looking for a recommendation for a USB ethernet device, any adapter with a `AX88179` chipset should work.

gabrielburnworth, 2022-01-26:
In addition to the above, from a quick review of firmware configs and sources:
```
Ralink rt2501/rt61
Ralink rt73
Ralink rt27xx/rt28xx/rt30xx
Realtek 8188/8192
Realtek 8712/8723/8723be
Realtek 8821a/8821c/8822b/8822cs/8822cu/8852au
```
configs:
https://github.com/FarmBot/farmbot_system_rpi3/blob/d3544d316998653e562990789e5ce5fc0d85598a/nerves_defconfig#L55-L60

sources:
https://github.com/buildroot/buildroot/blob/master/package/linux-firmware/linux-firmware.mk
https://github.com/buildroot/buildroot/blob/master/package/linux-firmware/Config.in

sabas1080, 2022-01-27:
Thanks @RickCarlino 

I have not been able to access any of those chips, what I have done is use an extender device like this one, if someone has the same problem I will leave the solution

https://www.tp-link.com/es/home-networking/range-extender/tl-wa855re/

closed 2022-01-27 by sabas1080

RickCarlino, 2022-01-27:
Thanks, @sabas1080 I have seen some customers use TPLink extender products in the past, though I cannot confirm the exact model number. If it is exposed as a standard WiFi AP, it should work. How is that device working out so far?