FarmBot/farmbot_os #1493: Boot Sequence execution may be delayed by earlier (overdue) FarmEvent sequence executions

jsimmonds2, 2022-04-30:
For the case where FBOS has been down (powered off, e.g.) for more than 15 minutes and an existing Sequence Farm Event execution becomes **_due_** within a time window starting 15 minutes prior to boot up, then the overdue Farm Event sequence will run before the Boot Sequence (see pic.)

I'd like to work on a solution but first I have questions :)
* Does FBOS reaally need to deal with overdue executions scheduled 15 minutes in the past ?
  (@RickCarlino was surprised the code did this)
* The CeleryScript Scheduler begins work on these overdue executions even well before the firmware is ready so they fail if they need the firmware !
  What is the desired user story here ?

Thanks :slightly_smiling_face: 

[image]

roryaronson, 2022-05-05:
@jsimmonds2 I think handling the overdue executions was originally to help users who scheduled sequence events too closely together. For example, if a sequence was scheduled at 7:20 took 1 minute and 3 seconds to complete, that would cause a sequence scheduled at 7:21 to get skipped. However, 15 minutes is an excessively long grace period I think. We could reduce that period to 1 minute, or remove that functionality altogether if it allows FBOS test coverage and overall code maintainability to increase. Any affected users will simply need to space their sequence executions farther apart or combine the back-to-back executions into a single sequence to avoid having to pay attention to how long things take to finish.

Any executions that were scheduled to occur while the device is powered off (or not fully booted) should be thrown out to avoid runtime errors (lack of firmware) and just because that is somewhat unexpected behavior in my opinion.

Thanks for looking into this @jsimmonds2!

jsimmonds2, 2022-05-05:
> Thanks for looking into this @jsimmonds2!

@roryaronson . . and thanks for **your** informative reaction !

Will submit a PR, based on this, very soon.